Monitor-My-Solar/monitormysolar #41: [HA Feature] Confirm whether invreboot and resetsetting should both appear as Restart Inverter in HA

danielraffel, 2024-11-20:
### Feature Name

Confirm whether invreboot and resetsetting should both appear as Restart Inverter in HA

### Is this an Entity?

No

### Feature Description

Confirm that these settings are setting the correct name in Home Assistant for their behavior. They both currently appear as `Restart Inverter` but I suspect they might do different things?
 
```
invreboot
resetsetting
```
[image]

### Home Assistant Version

2024.11.2

wrighpat, 2025-05-19:
Any feedback on this topic?

zakery292, 2025-05-19:
I'll take a look tomorrow and let you know