Monitor-My-Solar/monitormysolar #43: [HA Feature] new entity for calculating house consumption per day

danielraffel, 2024-11-22:
### Feature Name

new entity for calculating house consumption per day

### Is this an Entity?

No

### Feature Description

Currently, I use einv_day to estimate household consumption per day but I’m under the impression it’s not very accurate. It would be awesome to have a more accurate entity exposed in HA for house consumption per day (and not to mess with creating utility helpers in HA if not necessary.) also fine if it needs to be calculated in HA with a utility / helper.

### Home Assistant Version

2024.11.2

danielraffel, 2024-12-14:
🔌🔋🏡b-u-m-p😉

zakery292, 2025-04-02:
this is in 2.0.0 to be released and dongle FW 2.5.5 >

closed 2025-04-02 by zakery292

HPN365, 2025-05-15:
@zakery292 I'm on firmware 2.6.5 and I'm not seeing this in Home Assistant. What is the sensor called? I see Energy to user sensors but the all sensor appears to only be incrementing when there is solar power, aka during the night when the inverter is pulling from the battery the sensor does not increase. And the day sensor doesn't increase at all.

[image]

[image]

zakery292, 2025-05-15:
The new consumption sensors are available in the Beta version, this is nearly finished i just have not had time to finish it as of late, especially with the Gridboss integration that we are working on. 

I can provide the code to make the current version work with the new sensors if you like? if you can wait a few days ill release a temp update bumping the numbers slightly to include the new sensors.