Monitor-My-Solar/monitormysolar #46: [Dongle Feature] EG4 6000XP full integration.

rbaugh, 2024-11-29:
### Feature Name

Missing Data

### UI Or Firmware Feature?

Firmware

### Feature Description

When I was able to use SolarAssistant, I was able to get information for my PV MPPTs on the inverter. The only thing that I can get for PV values are the Power PV All and Total PV Day. But I am missing the current, volts, and total usage per mppt. 

I can see that these are available in the const.py file, but they are locked to certain firmware codes. The firmware code showing is CCAA. Oddly, when I look at the local connection on the dongle (web socket), and select the PV, I can see the current on each MPPT.

### Usage Scenarios

To get the data that I had with Solar Assistant and is available with the inverter.

### Current Firmware Version

2.1.4

zakery292, 2024-11-29:
what is your Firmware code? ill add it to the list

rbaugh, 2024-11-29:
CCAA

rbaugh, 2024-11-29:
I haven’t looked through the other sensors that are tied to the inverters to see if I may be missing others.

zakery292, 2024-11-29:
How many PV inputs? 2 or 3 ?

rbaugh, 2024-11-29:
2

zakery292, 2024-11-29:
Release added try that should give you everything. This firmware code was not on my list inverters. Apologies

zakery292, 2024-11-29:
There will be a dongle update for stability fixes around TLS connections in the next 24 hours just in the final stages of testing.

rbaugh, 2024-11-29:
Looks like you are missing a closing quote on line 65 in const.py

zakery292, 2024-11-29:
Thatll Teach me for editing on branch wont it. All done Apologies.

rbaugh, 2024-11-30:
Zak, sorry for the delayed response. I had to step away, and it happens to all of us.

I pulled down the changes, but not seeing the new entities. Should the CCAA be listed in the VALID_FIRMWARE_CODES? Not sure if that would matter or not. I also see you did add the colon as another commit, but the latest release doesn’t include it. Just FYI

Also, side question, but for my device. All of my temps are in celsius, how do I get them to Fahrenheit? I selected USA when configuring the dongle.

rbaugh, 2024-11-30:
Guess I should look closer at the code. I see the temps are listed in the const.py as celsius. So it answers part of my question. My lack of full HA knowledge, but is there a way that could be tied to an HA setting? I tried to force the entity in HA to use F but it just ignores it.

zakery292, 2024-11-30:
Ill have to do this on VS code tomorrow not GIT ill sort your sensors in the morning its 1am here i hope that's ok. 
Native Registers on the inverter are C not F there is no conversion on the dongle but that is a practical idea It'll be quicker if i can make HA do it so ill try that as well tomorrow. you might need to delete the integration and reload it or restart HA. Anytime you pull down new version you have to reboot or HA wont reload the entities, so it might be you need to restart for it to take effect please try that and let me know. 

Ill do the conversions tomorrow for temperature

rbaugh, 2024-11-30:
Thanks for responding so late.

I’ll try some more deletes and such. I tried clearing out the config files in `.storage` as well. Get some sleep and thanks for the assistance. I am getting started with the HA code side and also just trying to understand python. I am more c#/PHP

zakery292, 2024-11-30:
I hate HA its a nightmare. Ill make it work tomorrow. Good night!

rbaugh, 2024-12-01:
What’s the best way to see the data that the lux device is using?  The EPS L1/2 are not showing any values in HA/MQTT from the dongle. So I assume what you are using for the other inverters is slightly different on the SNA models.  The data being sent to the lux online UI is still correct. The combined/total usage seems correct just not getting the data for the individual EPS out lines. Also do you know if the PV amp per controller is available on the device?  I had it in SolarAssistant but not sure if they were reading it from the inverter or if it was equated

zakery292, 2024-12-01:
PV Amps is calculated.  this can be added and ill look to add that in the next release.  im just dealing with fault codes on the dongles and warnings then ill move over to adding those properly in to HA

All the data is there we might just need to set it in the consts for you. 

have you got a MQTT client you can run on your PC? 

use either MQTT FX or emxq client connnect to the broker and subscribe to DONGLEID/inputbankt1 && DONGLEID/inputbank2 

Look at the payloads, youll see all the data i drag out of the units and we can adjust from there.

zakery292, 2024-12-01:
Its very much possible im not parsing the data correctly on the dongle. i dont have these units accessible to me so im going off documentation i have which is a nightmare. 

There is a dongle update coming out today and HA update if you can add me in your input bank 1 and input bank2 payload data into this thread before the update ill take a look at my conversions

rbaugh, 2024-12-01:
Here is the data I was able to capture as close as possible in terms of the data that is showing in the EG4/Lux Data download and then pulling the MQTT data for the to input banks. 

LMK what else I can help with.

[3533740010 - 2024-12-01-3.xls](https://github.com/user-attachments/files/17969518/3533740010.-.2024-12-01-3.xls)
[inputBank1.json](https://github.com/user-attachments/files/17969519/inputBank1.json)
[inputBank2.json](https://github.com/user-attachments/files/17969520/inputBank2.json)

zakery292, 2024-12-01:
Looks like my registers might be off by one or two

zakery292, 2024-12-01:
Logically my register mapping is correct to the documentation.  I find this most perplexing. 

Im going to change the software on the update OTA update, the only way you will be able to update is by using MQTT and sending the command on topic 
DONGLEID/update
Command will be 
```
{
"firmware_update":"updatedongle"
}
```

Let me know when your ready. 

Once this is done the dongle will reboot, you will get a new topic. DONGLEID/unparsed  please can i have a copy of this. 
Let me know when your ready to update and ill upload the firmware

rbaugh, 2024-12-01:
So do you want me to still post the topic? Or is this going to be normal update?

zakery292, 2024-12-01:
Bare with me ive made a rod for my own back here Give me 5 minutes

zakery292, 2024-12-01:
Ive just been called into work ill be back on this ASAP

zakery292, 2024-12-01:
@rbaugh what dongle firmware are you on?

rbaugh, 2024-12-01:
2.1.4

rbaugh, 2024-12-01:
But for some reason now, I can't seem to get the UI to want to load. The MQTT data still seems to be coming in ok

zakery292, 2024-12-01:
Thank heavens don't worry about it give me a hour or two and I'll be home leave it till then

rbaugh, 2024-12-01:
Well, this is what is in the MQTT:
```
{
    "SW_VERSION": "2.1.1",
    "latestUiVersion": "2.0.60",
    "latestFirmwareVersion": "2.2.0",
}
```

zakery292, 2024-12-01:
Home now Uploading firmware shortly

rbaugh, 2024-12-01:
I just sat down to eat so will be distracted for a bit

zakery292, 2024-12-01:
Ok let me know when your ready and ill upload the firmware

rbaugh, 2024-12-01:
Are you available now or in a few minutes?

zakery292, 2024-12-01:
Im here, you should have a OTA update availble on the dongle.  if the UI is not working then reboot the dongle. then do the OTA, then do the UI 

you will then have a topic DONGLEID/unparsed 
i just need one response from that

rbaugh, 2024-12-01:
What's the operation to reboot? I saw your video to get the wifi, but that seemed like a reset rather than reboot. I still don't have the UI

rbaugh, 2024-12-01:
I have the dongle connect with the screws, so wasn't sure if there was a short press on the button rather than unscrewing to disconnect it.

zakery292, 2024-12-01:
either pull the dongle out and re insert it 
or a quick press of the button on the dongle

zakery292, 2024-12-01:
just a very quick press once

rbaugh, 2024-12-01:
It seems like the UI keeps locking up on me. I reset the dongle and tried to reload for the OTA and it hung up on me. I then tried to reboot again and it was as if the dongle wouldn’t reboot. Then it flashed and I am stuck trying to install from scratch. I can select the country and scan for WiFi, but then it hangs when trying to set the wifi

zakery292, 2024-12-01:
Hmm Very odd its hanging when connecting to wifi. 

Are you connected direct to the dongle? IOS or android ?

rbaugh, 2024-12-01:
Direct to dongle on my MacBook. When I apply the WiFi settings, it says to wait and just doesn’t do anything.

zakery292, 2024-12-01:
Oh thats good

zakery292, 2024-12-01:
You should at this point have no lights on the dongle and maybe when you reconnect you should be at a OTA page?

rbaugh, 2024-12-01:
No it stays with the red WiFi light.

zakery292, 2024-12-01:
Ughh. Try press and hold the button till all the lights flash its factory reset and try one more time. Poxy thing.

rbaugh, 2024-12-01:
Same thing. I get to saving WiFi settings and it hangs, or seems to hang. Lights on dongle never change, just the solid red WiFi light. Been about a min now waiting for it to save

zakery292, 2024-12-01:
The only thing i can suggest is to power the dongle on and off. Easiest way if its screwed in power off the inverter and on again. Sounds like its stuck. 
We just need to get it back on the internet then all this is solved. 
i dont understand why its being difficult

zakery292, 2024-12-01:
If it doesn't work i have some dongles in the US ill get one shipped out to you they are only in San Diego so ill get them updated and get it out

rbaugh, 2024-12-01:
So this is an odd one. I kept trying it and unplugged it. At one point after unplugging, I saw on my network the dongle was showing up on it. When I went to the IP, I got the set WiFi screen and was able to set and load 2.2.0

zakery292, 2024-12-01:
Thank heavens for that update the UI and you should now be all set. this should not do this any more i solved the memory leak in 2.2.0

zakery292, 2024-12-01:
Perfect thank you let me do some debugging

zakery292, 2024-12-01:
Could i please get the inputbank2 that comes at the same time as this payload please. Sorry to be a pain. 
The issue looks to be a parsing error but i cant see where. Logically its correct. the first two are voltages, second two are wattages the request is fine the data being returned is fine and in the correct order. 
there seems to be some confused on the mapping but i could not tell you where i need to see where they get lost from each other. looking at the original lot you sent it looks like they are one/two registers out but its not possible if that data set is being passed over

rbaugh, 2024-12-01:
I am seeing a lot of missing items at the moment. PV values and such but maybe that is intended.

I tried to get things as close as I could, but with MQTT explorer I have to select/copy/paste each and I am not that quick. This is raw, bank2, and then bank1. Sending extra to see if it helps any:
```
{"raw_data":[1197,1199,14,7,14,10,19,22,7246,0,6532,0,0,0,0,65392,0,0,0,0,0,0,0,126,0,126,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
{
	"Serialnumber":	"XXX",
	"payload":	{
		"EPSVoltL1N":	1197,
		"EPSVoltL2N":	119,
		"Peps_L1N":	1,
		"Peps_L2N":	0,
		"Seps_L1N":	1,
		"Seps_L2N":	10,
		"EepsL1N_day":	19,
		"EepsL2N_day":	22,
		"EepsL1N_all":	0,
		"EepsL2N_all":	0,
		"Qinv":	0,
		"AFCI_CurrCH1":	7268,
		"AFCI_CurrCH2":	0,
		"AFCI_CurrCH3":	0,
		"AFCI_CurrCH4":	0,
		"AFCI_ArcCH1":	0,
		"AFCI_ArcCH2":	0,
		"AFCI_ArcCH3":	0,
		"AFCI_ArcCH4":	0,
		"AFCI_MaxArcCH1":	0,
		"AFCI_MaxArcCH2":	0,
		"AFCI_MaxArcCH3":	0,
		"AFCI_MaxArcCH4":	0,
		"ACCouplePower":	0,
		"Pload170":	0,
		"Eload_day":	0,
		"Eload_all":	0,
		"EPS_overload_ctrl_time":	0,
		"Pinv_S":	0,
		"Pinv_T":	0,
		"Prec_S":	0,
		"Prec_T":	0,
		"Ptogrid_S":	0,
		"Ptogrid_T":	0,
		"Ptouser_S":	0,
		"Ptouser_T":	0,
		"GenPower_S":	0,
		"GenPower_T":	0,
		"IinvRMS_S":	0,
		"IinvRMS_T":	0,
		"PF_S":	0,
		"GridVoltL1N":	0,
		"GridVoltL2N":	0,
		"GenVoltL1N":	0,
		"GenVoltL2N":	0,
		"PinvL1N":	0,
		"PinvL2N":	0,
		"PrecL1N":	0,
		"PrecL2N":	0,
		"Ptogrid_L1N":	0,
		"Ptogrid_L2N":	0,
		"Ptouser_L1N":	0,
		"Ptouser_L2N":	0,
		"PF_T":	0,
		"ArcAlarmCH1":	0,
		"ArcAlarmCH2":	0,
		"ArcAlarmCH3":	0,
		"ArcAlarmCH4":	31457,
		"SelfTestResultCH1":	17443,
		"SelfTestResultCH2":	0,
		"SelfTestResultCH3":	0,
		"SelfTestResultCH4":	0
	}
}
{"serialnumber":"XXX","data":{"State":"0xC0","StateDescription":"Off Grid mode, PV and battery connected to grid","Vpv1":278,"Vpv2":276,"Vpv3":392,"Vbat":53,"SOC":99,"SOH":0,"Pload":0,"InternalFault":0,"Ppv1":51,"Ppv2":48,"Ppv3":0,"Pall":99,"Pcharge":13,"Pdischarge":0,"VacR":0,"VacS":0,"VacT":614.5,"Fac":0,"Pinv":0,"Prec":0,"IinvRMS":1,"PF":0,"VepsR":240.10000610351562,"VepsS":1004.7999877929688,"VepsT":2053,"Feps":60.020000457763672,"Peps":21,"Seps":22,"Ptogrid":0,"Ptouser":0,"Epv1_day":3.7000000476837158,"Epv2_day":3.5999999046325684,"Epv3_day":0,"Epv_all":7.3000001907348633,"Einv_day":0,"Erec_day":0,"Echg_day":3.4000000953674316,"Edischg_day":1.5,"Eeps_day":4,"Etogrid_day":0,"Etouser_day":0,"Vbus1":393,"Vbus2":373.20001220703125,"SW_VERSION":"2.1.5","latestUiVersion":"2.0.60","latestFirmwareVersion":"2.2.0","firmwareChangelog":"UI: Fixed offline inverters Error when loading page, fixed the Wi-Fi & MQTT icon tool tip, fixed ipad view, increased max sockets for further reliability and stability. \r\nFW: Fixed memory leak, Fixed SSL and TLS issues around FW updates (i think), Stability fixes, MQTT reconnection for TLS sessions, Corrected clean up on TLS sessions. Added logic for Cellular dongles","Epv1_All":1019,"Epv2_All":792,"Epv3_All":0,"Einv_All":0,"Erec_All":0,"Echg_All":525.5999755859375,"Edischg_All":395.29998779296875,"Eeps_All":1378.199951171875,"Etogrid_All":0,"Etouser_All":1019,"FaultCode":0,"WarningCode":0,"Tinner":35,"Tradiator1":35,"Tradiator2":37,"Tbat":36,"Reserved_69":0,"RunningTime":2359296,"wAutoTestLimit":0,"uwAutoTestDefaultTime":0,"uwAutoTestTripValue":0,"uwAutoTestTripTime":0,"ACInputType":0,"ACCoupleInverterFlow":0,"ACcoupleEn":0,"MaxChgCurr":15,"MaxDischgCurr":400,"ChargeVoltRef":56,"DischgCutVolt":0,"BatStatus0_BMS":0,"BatStatus5_BMS":4608,"BatStatus_INV":3,"BatParallelNum":4,"BatCapacity":400,"BatCurrent_BMS":0,"FaultCode_BMS":0,"WarningCode_BMS":0,"MaxCellVolt_BMS":3.3340001106262207,"MinCellVolt_BMS":3.3239998817443848,"MaxCellTemp_BMS":24,"MinCellTemp_BMS":22,"CycleCnt_BMS":25,"BatVoltSample_INV":53.299999237060547,"T1":0,"T2":0,"T3":0,"T4":0,"MasterOrSlave":260,"OnGridloadPower":0,"VBusP":194,"GenVolt":0,"GenFreq":0,"GenPower":32391,"Egen_day":512,"Egen_all":5536961},"events":{"state":{"value":"0xC0","description":"Off Grid mode, PV and battery connected to grid","timestamp":"2024-12-01T23:21:59Z"},"fault":{"value":0},"warning":{"value":0}}}
```

zakery292, 2024-12-01:
change your init.py to this 

```
import asyncio
import json
import logging
from homeassistant.core import HomeAssistant
from homeassistant.const import Platform
from homeassistant.helpers.event import async_call_later
from homeassistant.components import mqtt
from .const import DOMAIN, ENTITIES
from .mqttHandeler import MQTTHandler

_LOGGER = logging.getLogger(__name__)

async def async_setup_entry(hass: HomeAssistant, entry):
    try:
        _LOGGER.info(f"Setting up Monitor My Solar for {entry.data.get('inverter_brand')}")

        config = entry.data
        inverter_brand = config.get("inverter_brand")
        dongle_id = config.get("dongle_id")
        firmware_code = config.get("firmware_code")

        # Initialize the MQTT handler and store it in the hass data under the domain
        mqtt_handler = MQTTHandler(hass)
        hass.data.setdefault(DOMAIN, {})["mqtt_handler"] = mqtt_handler

        brand_entities = ENTITIES.get(inverter_brand, {})
        if not brand_entities:
            _LOGGER.error(f"No entities defined for inverter brand: {inverter_brand}")
            return False

        # Initialize the DOMAIN key in hass.data if it doesn't exist
        if DOMAIN not in hass.data:
            hass.data[DOMAIN] = {}

        async def process_incoming_message(msg):
            if msg.topic == f"{dongle_id}/firmwarecode/response":
                _LOGGER.debug("Received firmware code response")
                try:
                    data = json.loads(msg.payload)
                    firmware_code = data.get("FWCode")
                    if firmware_code:
                        hass.data[DOMAIN]["firmware_code"] = firmware_code
                        _LOGGER.debug(f"Firmware code received: {firmware_code}")
                        hass.config_entries.async_update_entry(
                            entry, data={**entry.data, "firmware_code": firmware_code}
                        )
                        setup_success = await setup_entities(hass, entry, inverter_brand, dongle_id, firmware_code)
                        if not setup_success:
                            return False
                    else:
                        _LOGGER.error("No firmware code found in response")
                except json.JSONDecodeError:
                    _LOGGER.error("Failed to decode JSON from response")
            elif msg.topic == f"{dongle_id}/response":
                await mqtt_handler.response_received(msg)
            elif msg.topic == f"{dongle_id}/status":
                await process_status_message(hass, msg.payload, dongle_id)
            else:
                await process_message(hass, msg.topic, msg.payload, dongle_id, inverter_brand)

        await mqtt.async_subscribe(hass, f"{dongle_id}/#", process_incoming_message)

        if not firmware_code:
            _LOGGER.debug("Requesting firmware code...")
            await mqtt.async_publish(hass, f"{dongle_id}/firmwarecode/request", "")

            async def firmware_timeout(_):
[... 178 lines cut ...]
    brand_entities = ENTITIES.get(inverter_brand, {})
    if not brand_entities:
        _LOGGER.debug(f"No entities defined for inverter brand: {inverter_brand}. Defaulting to 'sensor'.")
        return "sensor"

    entity_id_suffix_lower = entity_id_suffix.lower()
    _LOGGER.debug(f"Looking for entity_id_suffix '{entity_id_suffix_lower}' in brand '{inverter_brand}'.")

    for entity_type in ["sensor", "switch", "number", "time", "time_hhmm", "button", "select"]:
        if entity_type in brand_entities:
            for bank_name, entities in brand_entities[entity_type].items():
                _LOGGER.debug(f"Checking in bank '{bank_name}' for entity_type '{entity_type}'.")
                for entity in entities:
                    unique_id_lower = entity["unique_id"].lower()
                    _LOGGER.debug(f"Comparing with unique_id '{unique_id_lower}' for entity_type '{entity_type}'.")
                    if unique_id_lower == entity_id_suffix_lower:
                        if entity_type == "time_hhmm":
                            _LOGGER.debug(f"Matched entity_id_suffix '{entity_id_suffix_lower}' to entity type 'time'.")
                            return "time"
                        _LOGGER.debug(f"Matched entity_id_suffix '{entity_id_suffix_lower}' to entity type '{entity_type}'.")
                        return entity_type

    _LOGGER.debug(f"Could not match entity_id_suffix '{entity_id_suffix_lower}'. Defaulting to 'sensor'.")
    return "sensor"
```

zakery292, 2024-12-01:
The above init.py is the new one im working on to work with 2.2.0 and above its just worked on mine so it should work on yours

rbaugh, 2024-12-01:
Ok, trying it now. Stepped away for a moment.

rbaugh, 2024-12-01:
Entity was showing an error and this is in the logs:
Logger: custom_components.monitormysolar
Source: custom_components/monitormysolar/__init__.py:139
integration: MonitorMySolar ([documentation](https://github.com/zakery292/monitormysolar), [issues](https://github.com/zakery292/monitormysolar/issues))
First occurred: 17:45:15 (1 occurrences)
Last logged: 17:45:15

Error setting up update entities: No module named 'custom_components.monitormysolar.update'

zakery292, 2024-12-01:
its because of the platform 
```
platform update
```
Ive removed the offending code above and now should work

rbaugh, 2024-12-02:
We had a family get together today so I got pulled away cleaning up. I have things loaded without errors now, but the data is still off some. My PV is off atm so will check on it tomorrow. But not seeing individual L/L2 values. Same as before, raw, bank2, bank1.

I know it's late for you so we can pick up tomorrow unless you are on a roll.

```
{"raw_data":[1194,1197,161,114,161,114,21,22,7248,0,6532,0,0,0,0,0,0,0,0,0,0,0,0,126,0,126,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}
{
	"Serialnumber":	"XXX",
	"payload":	{
		"EPSVoltL1N":	1194,
		"EPSVoltL2N":	119,
		"Peps_L1N":	16,
		"Peps_L2N":	11,
		"Seps_L1N":	16,
		"Seps_L2N":	114,
		"EepsL1N_day":	21,
		"EepsL2N_day":	22,
		"EepsL1N_all":	0,
		"EepsL2N_all":	0,
		"Qinv":	0,
		"AFCI_CurrCH1":	7270,
		"AFCI_CurrCH2":	0,
		"AFCI_CurrCH3":	0,
		"AFCI_CurrCH4":	0,
		"AFCI_ArcCH1":	0,
		"AFCI_ArcCH2":	0,
		"AFCI_ArcCH3":	0,
		"AFCI_ArcCH4":	0,
		"AFCI_MaxArcCH1":	0,
		"AFCI_MaxArcCH2":	0,
		"AFCI_MaxArcCH3":	0,
		"AFCI_MaxArcCH4":	0,
		"ACCouplePower":	0,
		"Pload170":	0,
		"Eload_day":	0,
		"Eload_all":	0,
		"EPS_overload_ctrl_time":	0,
		"Pinv_S":	0,
		"Pinv_T":	0,
		"Prec_S":	0,
		"Prec_T":	0,
		"Ptogrid_S":	0,
		"Ptogrid_T":	0,
		"Ptouser_S":	0,
		"Ptouser_T":	0,
		"GenPower_S":	0,
		"GenPower_T":	0,
		"IinvRMS_S":	0,
		"IinvRMS_T":	0,
		"PF_S":	0,
		"GridVoltL1N":	0,
		"GridVoltL2N":	0,
		"GenVoltL1N":	0,
		"GenVoltL2N":	0,
		"PinvL1N":	0,
		"PinvL2N":	0,
		"PrecL1N":	0,
		"PrecL2N":	0,
		"Ptogrid_L1N":	0,
		"Ptogrid_L2N":	0,
		"Ptouser_L1N":	0,
		"Ptouser_L2N":	0,
		"PF_T":	0,
		"ArcAlarmCH1":	0,
		"ArcAlarmCH2":	0,
		"ArcAlarmCH3":	0,
		"ArcAlarmCH4":	0,
		"SelfTestResultCH1":	0,
		"SelfTestResultCH2":	0,
		"SelfTestResultCH3":	0,
		"SelfTestResultCH4":	0
	}
}

{
    "serialnumber": "XXX",
    "data": {
        "State": "0x40",
        "StateDescription": "Off Grid mode, Battery Discharging",
        "Vpv1": 0,
        "Vpv2": 0,
        "Vpv3": 391,
        "Vbat": 53,
        "SOC": 97,
        "SOH": 0,
        "Pload": 0,
        "InternalFault": 0,
        "Ppv1": 0,
        "Ppv2": 0,
        "Ppv3": 0,
        "Pall": 0,
        "Pcharge": 0,
        "Pdischarge": 290,
        "VacR": 0,
        "VacS": 0,
        "VacT": 614.5,
        "Fac": 0,
        "Pinv": 0,
        "Prec": 0,
        "IinvRMS": 1.7000000476837158,
        "PF": 0,
        "VepsR": 240.19999694824219,
        "VepsS": 1004.7999877929688,
        "VepsT": 2053,
        "Feps": 59.9900016784668,
        "Peps": 276,
        "Seps": 276,
        "Ptogrid": 0,
        "Ptouser": 0,
        "Epv1_day": 3.7999999523162842,
        "Epv2_day": 3.7000000476837158,
        "Epv3_day": 0,
        "Epv_all": 7.5,
        "Einv_day": 0,
        "Erec_day": 0,
        "Echg_day": 3.4000000953674316,
        "Edischg_day": 1.7999999523162842,
        "Eeps_day": 4.3000001907348633,
        "Etogrid_day": 0,
        "Etouser_day": 0,
        "Vbus1": 391.20001220703125,
        "Vbus2": 369.20001220703125,
        "SW_VERSION": "2.1.5",
        "latestUiVersion": "2.0.60",
        "latestFirmwareVersion": "2.2.0",
        "firmwareChangelog": "UI: Fixed offline inverters Error when loading page, fixed the Wi-Fi & MQTT icon tool tip, fixed ipad view, increased max sockets for further reliability and stability. \r\nFW: Fixed memory leak, Fixed SSL and TLS issues around FW updates (i think), Stability fixes, MQTT reconnection for TLS sessions, Corrected clean up on TLS sessions. Added logic for Cellular dongles",
        "Epv1_All": 1019,
        "Epv2_All": 792,
        "Epv3_All": 0,
        "Einv_All": 0,
        "Erec_All": 0,
        "Echg_All": 525.5999755859375,
        "Edischg_All": 395.60000610351562,
        "Eeps_All": 1378.5,
        "Etogrid_All": 0,
        "Etouser_All": 1019.0999755859375,
        "FaultCode": 0,
        "WarningCode": 0,
        "Tinner": 25,
        "Tradiator1": 26,
        "Tradiator2": 27,
        "Tbat": 24,
        "Reserved_69": 0,
        "RunningTime": 2424832,
        "wAutoTestLimit": 0,
        "uwAutoTestDefaultTime": 0,
        "uwAutoTestTripValue": 0,
        "uwAutoTestTripTime": 0,
        "ACInputType": 0,
        "ACCoupleInverterFlow": 0,
        "ACcoupleEn": 0,
        "MaxChgCurr": 45,
        "MaxDischgCurr": 400,
        "ChargeVoltRef": 56,
        "DischgCutVolt": 0,
        "BatStatus0_BMS": 0,
        "BatStatus5_BMS": 4608,
        "BatStatus_INV": 3,
        "BatParallelNum": 4,
        "BatCapacity": 400,
        "BatCurrent_BMS": -6.3000001907348633,
        "FaultCode_BMS": 0,
        "WarningCode_BMS": 0,
        "MaxCellVolt_BMS": 3.3190000057220459,
        "MinCellVolt_BMS": 3.312000036239624,
        "MaxCellTemp_BMS": 24,
        "MinCellTemp_BMS": 23,
        "CycleCnt_BMS": 25,
        "BatVoltSample_INV": 53,
        "T1": 0,
        "T2": 0,
        "T3": 0,
        "T4": 0,
        "MasterOrSlave": 260,
        "OnGridloadPower": 0,
        "VBusP": 189,
        "GenVolt": 0,
        "GenFreq": 0,
        "GenPower": 38705,
        "Egen_day": 512,
        "Egen_all": 5536961
    },
    "events": {
        "state": {
            "value": "0x40",
            "description": "Off Grid mode, Battery Discharging",
            "timestamp": "2024-12-02T01:07:13Z"
        },
        "fault": {
            "value": 0
        },
        "warning": {
            "value": 0
        }
    }
}
```

zakery292, 2024-12-02:
im not sure what your issue is at this moment im working on that but i want to sort out HA so im in the middle of that. your data will be off and most of the entities for setting settings wont work due to a mess-up on my behalf of the new payload structure. 

Ill release a HA update shortly once i get it working correctly my end which will solve a number of issues not including the one you have. 

When i release the HA update there will be another dongle update so i can see what data is actually being passed to the parsing function on the dongle. i think there are the first two registers missing which is putting off the entire run but i don't understand why so thank you for baring with me ill get to the bottom of it asap

rbaugh, 2024-12-02:
Thanks for sticking with it. It is a different model and was the only one up until recently. With the EG4 12000XP out it should mimic this model more. My inverter has some carry over from the 18kpv model but not as robust. I can clean up the HA features my model doesn’t have. I know some for sure and I assume you have been able to see some as well.

[10 more replies]