Monitor-My-Solar/monitormysolar #66: [HA Feature] Battery Charge/Discharge Rate (Lithium)

bdtrocke, 2025-05-15:
### Feature Name

Lithium Charge/Discharge Rate

### Is this an Entity?

No

### Feature Description

In HA, I am seeing a slider for Lead Acid Charge/Discharge control in Amps but those appear to be 0 for me and when adjusting they don't change my charge/discharge settings. I am using closed loop communication lithium batteries if that is a different setting than lead acid. 

### Home Assistant Version

2025.5.1

zakery292, 2025-05-15:
Are you using Lithium or Lead Acid for your battery communication type? 

If the inverter is set to lithium you should be using 
```
number.dongle_40_4c_ca_4e_90_00_chargepowerpercentcmd
```
and 
```
number.dongle_40_4c_ca_4e_90_00_dischgpowerpercentcmd
``

The above are % Values there are others like 
```
number.dongle_40_4c_ca_4e_90_00_dischgcurr
```
and 
```
number.dongle_40_4c_ca_4e_90_00_chargecurr
```

Please can you tell me your Firmware code it might be if your using the Flexboss 21 we need to add the Firmware code to the allowed list for certain settings 

```
sensor.dongle_40_4c_ca_4e_90_00_fwcode
```

Where my Dongle_XX_XX ... is replace with your dongle ID for the correct settings its likely that i don't have the FW code set for your unit. its a 5 minute fix and as soon as i have it ill post the update const.py file as a reply for you to paste in while we deal with  V2.0 Beta for Parallels and GridBoss 

Thanks

bdtrocke, 2025-05-16:
The Inverter is set to Lithium. I am running the EG4 18kpv. 2 of them in Parallel. One has the master inverter has the monitormysolar dongle and the other has the OEM dongle. I recently purchased another monitormysolar dongle for the slave inverter. And apologies about the technical disconnect on the above entity IDs. Do I need to edit the const.py file to get those to show in HA? Currently I don't see any of the above entity IDs in HA. For clarification I want to be able to control the charge and discharge amperage of the lithium batteries. 

Firmware code is: FAAB

zakery292, 2025-05-16:
So I just checked the files that firmware code does exist and is set, so you should have the 4 previously mentioned entities available to you. 

When the inverter is set in lithium mode you will cause either the charge command/charge current 

Same for discharge. 

I don't know how well you know home Assisant but if you go to 

Home Assisant >>> developer tools >>> states 

You can search for entities there or also in 

Settings >>> devices and services >>> entities 

And search there. 

If you still struggle to find the two entities let me know but in my previous example you would replace the dongle_XX_XX_XX_XX_XX_XX  with your dongle id followed by _dischgcurr << this would be discharge current 
_chargecurr <<< this would be charge current

bdtrocke, 2025-05-16:
I did end up finding the _chargepowerpercentcmd and _dischgpowerpercentcmd both at 100%. I don't see _dischgcurr and  _chargecurr, but that must be because I am running in Lithium mode? I accidentally left semicolons in the MAC address instead of underscores. I made a few changes to _chargepowerpercentcmd, 50%, 25%, 10% and it didn't appear to change anything on the inverter side. On the inverter LCD "Bat charge current limit (A)" stays at 250a no matter the setting and I'm not noticing any other setting change on the invert.

zakery292, 2025-05-16:
By the time you get up ill leave a block of code to add them in to your const.py  i cant do it now ive got to go off and go to work but by the time you wake up ill have it here for you. 

you'll copy from here the code block in to const.py in the correct section and then restart HA and they will be there

zakery292, 2025-05-17:
Apologies for the delay. 

please go to const.py and update the number section with this block 

```
            "number": {
                "holdbank2": [
                    {"name": "Active Power", "type": "number", "unique_id": "ActivePowerPercentCMD", "unit": "PERCENT", "min": 0, "max": 100 , "mode": "slider"},
                    {"name": "Charge Power Rate", "type": "number", "unique_id": "ChargePowerPercentCMD", "unit": "PERCENT", "min": 0, "max": 100, "mode": "slider"},
                    {"name": "Discharge Power Rate", "type": "number", "unique_id": "DischgPowerPercentCMD", "unit": "PERCENT", "min": 0, "max": 100 , "mode": "slider"},
                    {"name": "AC Charge Rate", "type": "number", "unique_id": "ACChgPowerCMD", "unit": "PERCENT", "min": 0, "max": 100, "mode": "slider"},
                    {"name": "AC Charge SOC Limit", "type": "number", "unique_id": "ACChgSOCLimit", "unit": "PERCENT", "min": 0, "max": 100, "mode": "slider", "class": "BATTERY"},
                    {"name": "Charge First Rate", "type": "number", "unique_id": "ChgFirstPowerCMD", "unit": "PERCENT", "min": 0, "max": 100, "mode": "slider"},
                    {"name": "Charge First SOC Limit", "type": "number", "unique_id": "ChgFirstSOCLimit", "unit": "PERCENT", "min": 0, "max": 100, "mode": "slider"},
                ],
                "holdbank3": [
                    {"name": "Force Discharge Power Rate", "type": "number", "unique_id": "ForcedDischgPowerCMD", "unit": "PERCENT", "min": 0, "max": 100 , "mode": "slider"},
                    {"name": "Force Discharge SOC Limit", "type": "number", "unique_id": "ForcedDischgSOCLimit", "unit": "PERCENT", "min": 0, "max": 100 , "mode": "slider"},
                    {"name": "Discharge Current DC(A)", "type": "number", "unique_id": "DischgCurr", "unit": "A", "min": 0, "max": 140, "mode": "slider"},

                    {"name": "Charge Current DC(A)", "type": "number", "unique_id": "ChargeCurr", "unit": "A", "min": 0, "max": 140, "mode": "slider"},
                    {"name": "Lead acid Charge Rate (A)", "type": "number", "unique_id": "ChargeRate", "unit": "A", "min": 0, "max": 140 , "mode": "slider", "native_unit": "A", "class": "CURRENT"},
                    {"name": "Lead Acid Discharge Rate (A)", "type": "number", "unique_id": "DischgRate", "unit": "A", "min": 0, "max": 140 , "mode": "slider", "native_unit": "A", "class": "CURRENT"},
                    {"name": "Battery Discharge Start Point (W)", "type": "number", "unique_id": "PtoUserStartdischg", "unit": "W", "min": 1, "max": 50 , "mode": "slider", "native_unit": "W", "class": "POWER"},
                    {"name": "Battery Charge Start Point (W)", "type": "number", "unique_id": "PtoUserStartchg", "unit": "W", "min": 1, "max": -50 , "mode": "slider", "native_unit": "W", "class": "POWER"},
                    {"name": "CT Offset (W)", "type": "number", "unique_id": "wCT_PowerOffset", "unit": "W", "min": 0, "max": 1000 , "mode": "slider", "native_unit": "W", "class": "POWER"},
                    {"name": "Export Power (%)", "type": "number", "unique_id": "MaxBackFlow", "unit": "W", "min": 0, "max": 200 , "mode": "slider", "state_class": SensorStateClass.MEASUREMENT, "unit_of_measurement": PERCENTAGE},
                    {"name": "On-grid Discharge Cut-off SOC Limit", "type": "number", "unique_id": "EOD", "unit": "PERCENT", "min": 0, "max": 90, "mode": "slider"},
                ],
```

bdtrocke, 2025-05-17:
Got it. Works great! Thanks!

closed 2025-05-17 by bdtrocke