Monitor-My-Solar/monitormysolar #68: [BUG] Entities do not exist for firmware code ceaa

wrighpat, 2025-05-27:
### Affected Component

Home Assistant

### Version Information

HA 1.3.2

### Media Attachments

[image]

[image]

[image]

### MQTT Output

```shell
Not sure where to find dongleID/status
```

### Detailed Description

I'm missing numerous entities. I'm trying to implement your custom dashboard and it indicates many entities do not exist. I would expect this for some entities, as I have an Off-grid EG4 12000xp, but there are other I would expect to exist.

### Issues

- [x] I agree i have checked the Issues Tab to see if this already Exists

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

wrighpat, 2025-05-27:
Integration won't startup now.

Logger: homeassistant.setup
Source: setup.py:344
First occurred: 15:26:47 (1 occurrence)
Last logged: 15:26:47

Setup failed for custom integration 'monitormysolar': Unable to import component: cannot import name 'LOGGER' from 'custom_components.monitormysolar.const' (/config/custom_components/monitormysolar/const.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 993, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1053, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/monitormysolar/__init__.py", line 2, in <module>
    from .const import LOGGER
ImportError: cannot import name 'LOGGER' from 'custom_components.monitormysolar.const' (/config/custom_components/monitormysolar/const.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 344, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1013, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1005, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1053, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/monitormysolar/__init__.py", line 2, in <module>
    from .const import LOGGER
ImportError: cannot import name 'LOGGER' from 'custom_components.monitormysolar.const' (/config/custom_components/monitormysolar/const.py)

zakery292, 2025-05-28:
There is nothing wrong with firmware. All the entities that are for your unit should now be available. I've got some more work to do on V2 it's in beta which won't work with version 1.4.4 CONST.py 

Between me and @rbaugh there is no one else developing this so updates take time as you can imagine. Epically as I have the dongle, web portal, mobile app and this all to do 😂 v2 is essentially the same just changes the way we do some things to make the code more up to date. V2 needs updating with your inverter also. Which again will be done in due course before release :) but in theory all the previously disabled entities should now work with the above const.py code