Monitor-My-Solar/monitormysolar #81: Wild erronous AC coupled power and energy readings

QuantumRipple, 2025-07-08:
### Affected Component

Dongle

### Version Information

3.0.0

### Media Attachments

[image]

### MQTT Output

```shell
N/A
```

### Detailed Description

Dongle Firmware 3.0.0
EG4 12000XP inverter
SmartLoad/AC coupled port on the inverter not wired to anything and disabled in the configuration.

Dongle direct web Power Flow view reports high AC coupled wattage (and associated energy) values when it should always be 0 or not even present.

### Issues

- [x] I agree i have checked the Issues Tab to see if this already Exists

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

QuantumRipple, 2025-07-10:
The HA integration seems to be reporting these erroneous values "Energy Of Generator" Today / All. The generator inputs to my inverter are likewise not connected to anything.

rbaugh, 2025-07-19:
This seems to also be for the EG4 6000XP. I get values for the “Energy of Generator All/Today” but the Generator voltages and AC Coupled power are always 0 as I don’t have anything connected to Grid nor Generator.

zakery292, 2025-07-29:
This is a LUX thing they seem to use un used registers on different models for debugging values. really annoying my side as i don't know what's real and what's not. if you don't use them best to disable the entity its impossible for me to what it genuine and what is not dynamically. HA Does not allow for the Dynamic hiding/removal of entities based on anther entities values to my knowledge. 

I can try and do this in the dongle before the data is sent EG if Gen is off force values to 0 if you would prefer, but this might create some undesired side effects down the line? 

The power flow in the dongle is changing to the new power flow shortly so again in the next few releases this will change and the power flow updated in due course

QuantumRipple, 2025-08-01:
If there's no register that indicates whether the AC coupled input is enabled (AC coupling is disabled in my 12000XP's settings), then a setting in the dongle to suppress that invalid power readout would be good.

There's a separate issue where the dongle is displaying these invalid values as AC coupled power/energy, but sends it to HA as generator power/energy. As it happens I _do_ have a generator connected now, which is a totally separate physical connection from the AC coupled input on the 12000XP (unlike the 6000XP where they were shared).

zakery292, 2025-08-01:
So as previously mentioned, as all entities are created on start I have no way of dynamically creating entities to appear and disappear on the fly within HA I've done my best with the profiles but as this is open source I hoped the community as they found bits missing or needed tweaking they would make adjustments just like @rbaugh has which I'm very very grateful for. 

I could dynamically set Ac coupled to 0 on the dongle based on settings values and the same with others but this poses a separate issue that if for what ever reason the logic i impose upon your current FW changes with a new FW release from EG4 then it might impose a negative effect which is why I assumed irrelevant entities the end user within HA would just disable them. 

I've provided access to 99% of registers they are within the payloads like the AC first registers but as there was no use at the time they are not parsed to write on the dongle (this is fixed in 3.1.0 along with the weird WiFi bug which is going out today) 

So I can't help that the manufacturer does weird things with registers that are currently not in use but I'll do everything I can. I'll try and add an option in the dongle config where if certain parameters are not in use the dongle will 0 them out. This will be trial and error as I don't physically have a 12 or a 6k unit to develop on issues have to be raised and I'll get to it when I can. HA is free and I allocate as much time to it as I can to mitigate the fact I can't respond or update HA instantly it's open source with free access to changes for you guys to make and I'll look at the pull requests run some tests and push your code.

I'm trying my best to keep everyone happy improvements on the dongle, new UI on the dongle greater stability and parallel ability within HA itself and @rbaugh's re write. There is only one of me running this whole thing currently. I promise I'll do my best which is all I can do

zakery292, 2025-08-02:
@QuantumRipple  please provide me your <dongleid>/inputbank1 & <dongleid>/inputbank2 payloads please. Once i have this i can debug further and see whats going on compared to others with your unit. or connect to Monitor My Solar and ill get the payloads from there myself let me know what you want to do.

QuantumRipple, 2025-08-03:
Here are the MQTT payloads @zakery292 :
```
/inputbank1 (qos=0): b'{"serialnumber":"44830P0464","payload":{"State":"0x11","StateDescription":"Unknown State","Vpv1":0,"Vpv2":0,"Vpv3":0,"Vbat":52,"SOC":64,"SOH":0,"Pload":1507,"InternalFault":0,"Ppv1":0,"Ppv2":0,"Ppv3":0,"Pall":0,"Pcharge":0,"Pdischarge":0,"VacR":249.69999694824219,"VacS":0,"VacT":3891.300048828125,"Fac":60,"Pinv":0,"Prec":0,"IinvRMS":0,"PF":0,"VepsR":247.39999389648438,"VepsS":872.0999755859375,"VepsT":420.79998779296875,"Feps":60.0099983215332,"Peps":0,"Seps":0,"Ptogrid":0,"Ptouser":1507,"HourlyConsumption":1419.106201171875,"DailyConsumption":14857.1591796875,"Epv1_day":15.199999809265137,"Epv2_day":0,"Epv3_day":0,"Epv_all":15.199999809265137,"Einv_day":0,"Erec_day":0,"Echg_day":6,"Edischg_day":7,"Eeps_day":14.699999809265137,"Etogrid_day":0,"Etouser_day":13.600000381469727,"Vbus1":1.7000000476837158,"Vbus2":0.5,"SW_VERSION":"3.0.0","UI_VERSION":"2.0.70","Epv1_All":128,"Epv2_All":0,"Epv3_All":0,"Einv_All":0,"Erec_All":76.6999969482422,"Echg_All":125.09999847412109,"Edischg_All":128.39999389648438,"Eeps_All":162.39999389648438,"Etogrid_All":0,"Etouser_All":994.20001220703125,"FaultCode":0,"WarningCode":0,"Tinner":0,"Tradiator1":40,"Tradiator2":44,"Tbat":0,"Reserved_69":0,"RunningTime":0,"wAutoTestLimit":0,"uwAutoTestDefaultTime":0,"uwAutoTestTripValue":0,"uwAutoTestTripTime":0,"ACInputType":0,"ACCoupleInverterFlow":0,"ACcoupleEn":0,"MaxChgCurr":60,"MaxDischgCurr":60,"ChargeVoltRef":57.4000015258789,"DischgCutVolt":0,"BatStatus0_BMS":0,"BatStatus5_BMS":12924,"BatStatus_INV":3,"BatParallelNum":6,"BatCapacity":600,"BatCurrent_BMS":0,"FaultCode_BMS":0,"WarningCode_BMS":0,"MaxCellVolt_BMS":3.2990000247955322,"MinCellVolt_BMS":3.2929999828338623,"MaxCellTemp_BMS":27,"MinCellTemp_BMS":26,"CycleCnt_BMS":0,"BatVoltSample_INV":52.799999237060547,"T1":0,"T2":0,"T3":0,"T4":0,"MasterOrSlave":260,"OnGridloadPower":0,"VBusP":0,"GenVolt":0,"GenFreq":0,"GenPower":61533,"Egen_day":153,"Egen_all":132627},"events":{"state":{"value":"0x11","description":"Unknown State","timestamp":"2025-08-03T02:47:05Z"},"fault":{"value":0},"warning":{"value":0}},"powerFlowDirection":{"value":0,"SmartLoadFlow":0,"SmartLoadEnOn":0,"EpsLoadPowerShow":0,"GridLoadPowerShow":0,"PloadPowerShow":0}}'
/inputbank2 (qos=0): b'{\n\t"Serialnumber":\t"44830P0464",\n\t"payload":\t{\n\t\t"EPSVoltL1N":\t124,\n\t\t"EPSVoltL2N":\t123,\n\t\t"Peps_L1N":\t0,\n\t\t"Peps_L2N":\t0,\n\t\t"Seps_L1N":\t0,\n\t\t"Seps_L2N":\t0,\n\t\t"EepsL1N_day":\t3,\n\t\t"EepsL2N_day":\t11,\n\t\t"EepsL1N_all":\t58,\n\t\t"EepsL2N_all":\t104,\n\t\t"Qinv":\t0,\n\t\t"AFCI_CurrCH1":\t1,\n\t\t"AFCI_CurrCH2":\t0,\n\t\t"AFCI_CurrCH3":\t0,\n\t\t"AFCI_CurrCH4":\t0,\n\t\t"AFCI_ArcCH1":\t0,\n\t\t"AFCI_ArcCH2":\t0,\n\t\t"AFCI_ArcCH3":\t0,\n\t\t"AFCI_ArcCH4":\t0,\n\t\t"AFCI_MaxArcCH1":\t0,\n\t\t"AFCI_MaxArcCH2":\t6,\n\t\t"AFCI_MaxArcCH3":\t0,\n\t\t"AFCI_MaxArcCH4":\t0,\n\t\t"ACCouplePower":\t0,\n\t\t"Pload170":\t0,\n\t\t"Eload_day":\t0,\n\t\t"Eload_all":\t0,\n\t\t"EPS_overload_ctrl_time":\t0,\n\t\t"Pinv_S":\t0,\n\t\t"Pinv_T":\t0,\n\t\t"Prec_S":\t0,\n\t\t"Prec_T":\t0,\n\t\t"Ptogrid_S":\t0,\n\t\t"Ptogrid_T":\t0,\n\t\t"Ptouser_S":\t0,\n\t\t"Ptouser_T":\t0,\n\t\t"GenPower_S":\t0,\n\t\t"GenPower_T":\t0,\n\t\t"IinvRMS_S":\t0,\n\t\t"IinvRMS_T":\t0,\n\t\t"PF_S":\t0,\n\t\t"GridVoltL1N":\t125,\n\t\t"GridVoltL2N":\t123,\n\t\t"GenVoltL1N":\t0,\n\t\t"GenVoltL2N":\t0,\n\t\t"PinvL1N":\t0,\n\t\t"PinvL2N":\t0,\n\t\t"PrecL1N":\t0,\n\t\t"PrecL2N":\t0,\n\t\t"Ptogrid_L1N":\t0,\n\t\t"Ptogrid_L2N":\t0,\n\t\t"Ptouser_L1N":\t391,\n\t\t"Ptouser_L2N":\t1123,\n\t\t"PF_T":\t0,\n\t\t"ArcAlarmCH1":\t0,\n\t\t"ArcAlarmCH2":\t0,\n\t\t"ArcAlarmCH3":\t0,\n\t\t"ArcAlarmCH4":\t0,\n\t\t"SelfTestResultCH1":\t0,\n\t\t"SelfTestResultCH2":\t0,\n\t\t"SelfTestResultCH3":\t0,\n\t\t"SelfTestResultCH4":\t0\n\t}\n}'
/status (qos=0): b'{"status":"alive","uptime":"12 20 22 55","freeheap":166200,"minfreeheap":130568,"largest_block":122856,"stackusage":1108,"HA_State_MQTT":"Connected","Web_State_MQTT":"Disconnected","Web_Error":"","HA_Error":"","Last_Reset_Reason":"Power-on reset"}'
/holdbank1 (qos=0): b'{"Serialnumber":"44830P0464","payload":{"FWCode":"ceaa","SlaveVer":0,"ComVer":4,"CntlVer":5,"FWVer":1,"ResetSetting":0,"time_Date_combined":"25-08-02 21:46:41","ComAddr":1,"Language":1,"DeviceType":54,"PVInputModel":4,"StartPVVolt":6.5,"ConnectTime":30,"ReconnectTime":60,"GridVoltConnLow":195.5,"GridVoltConnHigh":253,"GridFreqConnLow":47,"GridFreqConnHigh":50.5,"GridVoltLimit1Low":195.5,"GridVoltLimit1High":253,"GridVoltLimit1LowTime":50,"GridVoltLimit1HighTime":200,"GridVoltLimit2Low":115,"GridVoltLimit2High":264.5,"GridVoltLimit2LowTime":1,"GridVoltLimit2HighTime":10,"GridVoltLimit3Low":0,"GridVoltLimit3High":276,"GridVoltLimit3LowTime":0,"EPS":1,"OVFLoadDerate":0,"DRMS":0,"LVRT":0,"AntiIsland":0,"NeutralDetect":0,"GridOnPowerSS":0,"ACCharge":0,"SWSeamlessly":1,"SetToStandby":1,"ForcedDischg":0,"ForcedChg":0,"ISO":1,"GFCI":0,"DCI":0,"FeedInGrid":0,"BatteryType":2,"BatteryBrand":2,"BatteryCapacity":0,"PowerRating":6,"RuleMask":0,"RuleID":1}}'
/holdbank2 (qos=0): b'{"Serialnumber":"44830P0464","payload":{"PowerSoftStartSlope":100,"ActivePowerPercentCMD":100,"ChargePowerPercentCMD":100,"DischgPowerPercentCMD":100,"ACChgPowerCMD":100,"ACChgSOCLimit":0,"ACChgStart":"00:00","ACChgEnd":"00:00","ACChgStart1":"00:00","ACChgEnd1":"00:00","ACChgStart2":"00:00","ACChgEnd2":"00:00","ChgFirstPowerCMD":0,"ChgFirstSOCLimit":0,"ChgFirstStart":"00:00","ChgFirstEnd":"00:00","ChgFirstStart1":"00 [... cut ...]
I'm a little confused why HA shows the sensors for Generator energy at all, now that I see [const.py](https://github.com/Monitor-My-Solar/monitormysolar/blob/b2ee18cd9618872b41d95bc8390b9252583edb97/custom_components/monitormysolar/const.py#L337) does not list ceaa as an allowed device type for them.

Don't worry, I'm not expecting perfection right out the gate nor nor do I expect instant updates. I'm with you on the open source bit, that's why it's reported here so you know about it despite not having a 12000XP to test yourself.

QuantumRipple, 2025-08-03:
I actually did have the grid disconnected before and for the first half of my generator test. The gen _power_ registers from /inputbank2 don't feed HA entities, so I'm not sure what they were doing, if anything. 
`AC Coupled Power` from /inputbank2 remained 0 throughout.
The `Energy Of Generator Today` showed what I assume to be debug values that correlated with the working mode.

Here's some interesting captured data from the test:
[image]

`House Consumption (Live)` tracks `Power to User(live)` outside of when the generator is on and charging the batteries. Both are 0 when EPS is supplying house power.

I probably won't run another generator test until I get a better generator. While the 12000XP was fine with the traditional non-inverter generator's somewhat dirty power, the UPS units that protect my computers had an absolute fit , thrashing their relays and one shutting off prematurely.

QuantumRipple, 2025-08-03:
I'll go ahead and close this dongle power flow UI issue whenever the fixed firmware rolls out for retest.